morkai/h5.modbus #5: Publish to NPM

JeremyBYU, 2015-04-20:
Hey,
  I want to use this with MeteorJS.  Best way I see how to integrate this with Meteor is through NPM (https://github.com/meteorhacks/npm).  Can you publish this to NPM?

Thanks!

morkai, 2015-04-20:
According to the site you've linked, you can insert this into the `packages.json` file to use the package:
```json
{
  "h5.modbus": "https://github.com/morkai/h5.modbus/archive/0993906e41429a298e91c85dda00ea663c0e9f3e.tar.gz"
}
```

JeremyBYU, 2015-04-22:
Thanks,  got it working.  Will be using this for serial modbus but am testing with TCP Modbus now.  Quick question from the following code:

```
            t2 = master.readHoldingRegisters(0x0000, 8, {
                unit: 1,
                maxRetries: 10,
                timeout: 100,
                interval: 3000,
                onComplete: function(err, response) {
                    if (err) {
                        console.err(err.message);
        				console.err('I make the error here!');
                    } else {
                		console.log(response);
                		console.log(response.exceptionCode);
                        console.log(response.values.readUInt16BE(0));
                    }
                    

                }
            });
```
Notice that if my simulator is running (good connection) but there is a bad response (meaning an exception, maybe simulating wrong slave id). That the "err" variable is still empty, but that the response now will have the variable exceptionCode (doesn't exist if successful response).  Does this mean that if I really want to check for an error that I should be checking the existence of exception code? Mapping of the exceptionCode to meaning?

Also any method that will read response buffers into standard format, or do we have to do it ourselves.  

Thanks!

Jeremy

morkai, 2015-04-22:
> Noticed that if my simulator is running (good connection) but there is a bad response (meaning an exception, maybe simulating wrong slave id). That the "err" variable is still empty, but that the response now will have the variable exceptionCode (doesn't exist if successful response). Does this mean that if I really want to check for an error that I should be checking the existence of exception code? Any Mapping of the exceptionCode meaning?

`err`, if any, is an instance of `Error` and will be specified when no complete response was received from the slave (exception is still a valid response).

`res`, if any, is an instance of `h5.modbus.functions.Response`. Every response has `getCode()` and `isException()` method. If the slave returned an exception, the response will be an instance of `ExceptionResponse`, the `res.isException()` will return `true`, `res.getExceptionCode()` will be available and the `res.toString()` will return a readable exception description.

> Max retries doesn't seem to be honored in this case as well. Meaning if there's an exceptionCode (no buffer) it still keeps trying....

The `Master` has a configuration option `retryOnException` which is `true` by default.

> Also any method that will read response buffers into standard format (in other words not using readUInt16), or do we have to do it ourselves.

Everyone has their own standard format, so you're stuck with what the `Buffer` offers you or use existing `Buffer`-wrapping libraries or write your own helper functions for common use cases :)

For example, [here I have CSV list of tags](https://github.com/morkai/walkner-hydro/blob/master/config/tags.csv]) that [are converted to MODBUS transactions here](https://github.com/morkai/walkner-hydro/tree/master/backend/modules/modbus).

JeremyBYU, 2015-04-23:
Awesome! I will try this out. Great repository as well for the walkner-hydro.

Thanks,

Jeremy

closed 2019-10-09 by JeremyBYU