morkai/h5.modbus #10: modbus tcp connection interrups

micha1983, 2016-10-25:
hi,
i testet the h5.modbus module.
i started a read connection to a wago 750-841 plc.
nearly after 3min i get always a disconnected event triggered.

the interuption takes 200ms. the pings during this timeperiod are constantly nearly at 6ms.
has somebody also have seen thhis phanomen?

morkai, 2016-10-25:
Do you send any data during that time?

micha1983, 2016-10-25:
only once at startup one holding for testing.
could this be the problem, that i should constantly send something?

morkai, 2016-10-25:
Yes. The device you are using might be disconnecting you, because nothing is being sent and you're connected, so it might assume that the connection is broken.

micha1983, 2016-10-25:
thanks for the feedback. i will test this later. 
Great support!

micha1983, 2016-10-25:
ok. i've tested it and i'm writing at the moment every 10sec a value.
but i still habe the problems of an disconnect event.

morkai, 2016-10-25:
Show me some code and its results :)

micha1983, 2016-10-25:
ok. after my last post i disabled the powersave mode of my PC. at the moment i'm having a active connection since 20minutes. every second i'm writing one value

micha1983, 2016-10-25:
ok it was the energysaving mode of my PC. sorry.

closed 2016-10-25 by micha1983