morkai/h5.modbus #12: Connecting over TCP even though I choose serial

pakerfeldt, 2016-12-16:
I have been using 0.1.0 for a while but tried to bump to 1.0.0. However, now it seems to try to open a TCP connection even though I have specified a serial port.

This is how I create the master:
```javascript
var port = new SerialPort(serialPath, { baudrate: baudrate, parity: parity }, false);
  this.modbus = library.createMaster({
    transport: {
      type: 'rtu',
      eofTimeout: 500,
      connection: {
        type: 'serial',
        serialPort: port
      }
    },
    suppressTransactionErrors: false,
    retryOnException: true,
    maxConcurrentRequests: 1,
    defaultUnit: 0,
    defaultMaxRetries: 1,
    defaultTimeout: 1000,
  });
```
And the log prints:

> Master error { [Error: connect ECONNREFUSED 127.0.0.1:502]
  code: 'ECONNREFUSED',
  errno: 'ECONNREFUSED',
  syscall: 'connect',
  address: '127.0.0.1',
  port: 502 }

Am I missing something?

morkai, 2016-12-16:
The `connection` options object should now be specified in the master's options and not the transport's:
```js
var port = new SerialPort(serialPath, { baudrate: baudrate, parity: parity }, false);
  this.modbus = library.createMaster({
    transport: {
      type: 'rtu',
      eofTimeout: 500
    },
    connection: {
      type: 'serial',
      serialPort: port
    },
    suppressTransactionErrors: false,
    retryOnException: true,
    maxConcurrentRequests: 1,
    defaultUnit: 0,
    defaultMaxRetries: 1,
    defaultTimeout: 1000,
  });
```

pakerfeldt, 2016-12-16:
👍

closed 2016-12-16 by pakerfeldt