morkai/h5.modbus #13: The data address is not an allowable address for the server (or slave).

micha1983, 2017-01-02:
HI,

 i want to read the following modbus address over TCP.

ID 2
Modbusaddress: 30775

But everytime i get the following errormessage.
can somebofy help me with that?

0x03 (RES) Exception (2): Illegal Data Address: The data address received in the query is not an allowable address for the server (or slave).

with qModMaster.exe i'm able to read the address.
can i adjust the max allowed value somewhere?

thanks for the feedback.

Greetings
Michael

morkai, 2017-01-02:
Try
```js
const startingAddress = 30775;
const quantity = 1;
const unit = 2;

master.readHoldingRegisters(startingAddress, quantity, {unit: unit})
```

micha1983, 2017-01-02:
Thanks. Is working great.

closed 2017-01-02 by micha1983