morkai/h5.modbus #15: Revers connection RTU over TCP

goranach, 2017-03-29:
Is there way to poll such system:
PLC <-> 3G modem <-> PC?
But modem hasn't external ip, so modem init TCP connection to PC. And than start poll RTU.
http://www.irz.net/en/products/terminals/atm-series/atm3-485

morkai, 2017-03-30:
PC w/ external IP:

  - has a server that listens on port 3000, anything it receives should be forwarded to the client of the 502 server
  - has a local server that listens on port 502, anything it receives should be forwarded to the client of the 3000 server
  - has a Modbus Master that connects to the local server on port 502

PC w/ 3G modem:

  - connects to PC w/ external IP on port 3000, anything it receives should be forwarded to the PLC
  - connects to the PLC, anything it receives should be forwarded to the PC w/ external IP on port 3000

[Something like this.](https://gist.github.com/morkai/3c7f097dd35ea34981ca8fffc7511f1c)

goranach, 2017-03-30:
Thank you very much. Thought so.

closed 2017-06-03 by morkai