morkai/h5.modbus #17: Example for slave

Yarisrespect, 2017-04-18:
How to send request to the client?

I have following code:
```
listener.on('client', function(client){
    // how to send request?
});

const slave = new modbus.Slave({
    listener: listener,
    transport: transport,
    requestHandler: handleRequest
});
```
I'll be very thankful!

morkai, 2017-04-18:
Option `requestHandler` must be specified and be a `function(number unit, Request request, function(any) respond)`. For example:
```js
function handleRequest(unit, request, respond)
{
  if (isValidRequest(unit, request))
  {
    respond(new Buffer([1, 2, 3, 4]));
  }
  else
  {
    respond(ExceptionCode.IllegalDataAddress);
  }
}
```
`respond()` takes a single argument that can be one of:
  - `ExceptionCode` enum - will send an exception response
  - an instance of `Response` corresponding to the received `Request`
  - an instance of `Error` (more like an object with a `message` property) - will send the Slave Device Failure exception
  - an instance of `Buffer` - will try to create an instance of `Response` corresponding to the received `Request` by using the `Response.fromBuffer()` (for example, if the request is `ReadHoldingRegistersRequest`, then the `ReadHoldingRegistersResponse.fromBuffer()` will be called)
  - an object - will try to create an instance of `Response` corresponding to the received `Request` by using the `Response.fromOptions()`.

See [examples\slave.js](https://github.com/morkai/h5.modbus/blob/master/examples/slave.js) for a full example.

Yarisrespect, 2017-04-18:
What about transactions if it is tcp/ip?

morkai, 2017-04-18:
Use `TcpListener` and `IpTransport`.

Yarisrespect, 2017-04-18:
That's clear I mean executing concurrent transactions.

morkai, 2017-04-18:
The `respond()` callback is unique to each request. If you receive 10 requests, the `requestHandler` will be called 10 times, each with different `request` and `respond`.

Yarisrespect, 2017-04-18:
Thank you a lot! And last one question.
Am I correct?
To send command to the client I should send it in the request handler by calling a respond function?

morkai, 2017-04-18:
Yes, you must call `respond()` to send a response to a request.

Yarisrespect, 2017-04-18:
So, is modbus.Slave for emulating modbus device?

morkai, 2017-04-18:
Yes, modbus.Slave=server and modbus.Master=client.

closed 2017-06-03 by morkai