morkai/h5.modbus #23: Reason of 'data' event removal from h5.modbus

ekawahyu, 2017-12-05:
I am working on multiple Modbus devices using h5.modbus and virtual-serialport to form a virtual RS485 network on my computer. I found out that h5.modbus latest changes break data flow of virtual-serialport. Virtual-serialport expects to see 'data' event from h5.modbus in order to work. When I added back those 'data' event handler and binding within h5.modbus library, the virtual-serialport worked as expected. So, what was the reason for serialport 'data' event and binding removal in h5.modbus? Thank you.

morkai, 2017-12-05:
`SerialListener` and `SerialConnection` were updated to the latest version of the `serialport` package.

According to the [serialport's upgrade guide](https://github.com/node-serialport/node-serialport/blob/master/UPGRADE_GUIDE.md), the `readable` event is a preferred way of reading data over the `data` event (and in Node.js in general).

`serialport` is at version 6. `virtual-serialport` emulates `serialport` version <4.

ekawahyu, 2017-12-05:
Ok, got it! Thanks for the clarification!

closed 2017-12-05 by ekawahyu