morkai/h5.modbus #24: Slave Device IDs

ekawahyu, 2017-12-12:
I have some questions about connecting more than one slave devices to a master. Modbus protocol is RTU, so I have physical (and virtual) RS485 network for this. I took the example and modified the UNIT_TO_DATA object for slave ID=1 as this:
```javascript
const UNIT_TO_DATA = {
  0x01: {
    coils: new Array(0xFFFF),
    discreteInputs: new Array(0xFFFF),
    holdingRegisters: new Buffer(0xFFFF * 2).fill(0),
    inputRegisters: new Buffer(0xFFFF * 2).fill(0)
  }
};
```
And slave ID=2 like this:
```javascript
const UNIT_TO_DATA = {
  0x02: {
    coils: new Array(0xFFFF),
    discreteInputs: new Array(0xFFFF),
    holdingRegisters: new Buffer(0xFFFF * 2).fill(0),
    inputRegisters: new Buffer(0xFFFF * 2).fill(0)
  }
};
```
They both work just fine, but when the modbus request is sent to slave ID=1, slave ID=2 gets `Illegal Data Address` exception. And vice versa. In my opinion, any message comes with mismatched slave ID, it should just drop it and do not send any exception.

Another question, for master device, how to send a modbus request and pass the unit ID as parameter, so that it does not pick up the `defaultUnit`?

morkai, 2017-12-13:
Why do you have two different `UNIT_TO_DATA` variables?

The `Slave` doesn't send the `Illegal Data Address` exception. You choose when it's sent in the `requestHandler`.

You can specify custom unit along with other options when calling a request function, for example:
```js
master.readHoldingRegisters(0, 1, {unit: 1});
```

ekawahyu, 2017-12-13:
Because they are on two separate computers, 0x01 in one desktop and another as 0x02. The third computer is the master. They all are on a wired RS485 network.

morkai, 2017-12-13:
So if you're using the example, you should remove [this line](https://github.com/morkai/h5.modbus/blob/master/examples/slave.js#L174).

ekawahyu, 2017-12-13:
Ok, got it, thanks!

closed 2017-12-13 by ekawahyu