morkai/h5.modbus #25: Write to Holding Registers

ekawahyu, 2017-12-28:
Is there example to write to Holding Registers?

morkai, 2017-12-28:
```js
var address = 0x0000;
var value = 0; // (u)int16 or Buffer(2)
master.writeSingleRegister(address, value, options);

var startingAddress = 0x0000;
var values = [0, 1, 2, 3]; // an array of n (u)int16s or Buffer(n*2) where n is a number of registers to write
master.writeMultipleRegisters(startingAddress, values, options);
```

morkai, 2017-12-28:
Specific exception code:
```js
var exceptionCode = ExceptionCode.IllegalFunctionCode;
respond(exceptionCode);
```
Same as `ExceptionCode.SlaveDeviceFailure`:
```js
var err = new Error('message');
respond(err);
```
Response created from an options object and the request's function code:
```js
var options = {
  address: request.address,
  value: request.value
};
respond(options); // WriteSingleRegisterResponse.fromOptions(options) will be called
// or, if the request's properties match the response options:
respond(request);
```
Response created from a valid Buffer:
```js
var buffer = new modbus.WriteSingleRegisterResponse(request.address, request.value).toBuffer();
respond(buffer); // WriteSingleRegisterResponse.fromBuffer(buffer) will be called
```
Specific Response:
```js
var response = new modbus.WriteSingleRegisterResponse(request.address, request.value);
respond(response);
```

ekawahyu, 2018-01-02:
This simple test of printing out `.startingAddress` and `.values` are not working for `WriteMultipleRegisterRequest` handler:
```javascript
function HandleWriteMultipleRegistersRequest(unit, request, functionData, respond)
{
  console.log(request.startingAddress);
  console.log(request.values);

  //respond(cbExceptionCode.IllegalFunctionCode);
}
```
The starting address seems correctly received, but the values are `undefined` instead of showing the content of a buffer. What part am I missing?

ekawahyu, 2018-01-02:
I made a little modification to the data as follow:
```javascript
  var startingAddress = 0x0002;
  var values = [65, 66, 67, 68, 69, 70];
```
But the response is still the same:
```javascript
2
6
ABCDEF
0x10 (RES) Set register at address 2.
```

morkai, 2018-01-02:
[I've commited](https://github.com/morkai/h5.modbus/blob/c051b6cbce4ae6c8cc81f9989081f5b773d19c87/examples/slave.js#L339) an example implementation of the write single register & write multiple registers functions.

Can be tested with:
```js
'use strict';

const modbus = require('h5.modbus');

const master = modbus.createMaster({defaultUnit: 1});

master.on('error', err => console.log('[error]', err.message));
master.on('open', () => console.log('[open]'));
master.on('close', () => console.log('[close]'));

master.once('open', () =>
{
  read();
  setTimeout(writeSingle, 1337);
  setTimeout(writeMultiple, 1337 * 2);
});

function read()
{
  const tRead = master.readHoldingRegisters(0xFFFF - 9, 10, {
    interval: 1000
  });

  tRead.on('complete', onComplete);
}

function writeSingle()
{
  master.writeSingleRegister(0xFFFF, 0xABCD, onComplete);
}

function writeMultiple()
{
  const data = [
    0x1111, 0x2222, 0x3333, 0x4444, 0x5555, 0x6666, 0x7777, 0x8888, 0x9999, 0x0000
  ];

  master.writeMultipleRegisters(0xFFFF - 9, data, onComplete);
}

function onComplete(err, res)
{
  if (err)
  {
    console.log(`${this.id}: ${err.message}`);
  }
  else
  {
    console.log(`${this.id}: ${res}`);
  }
}
```

ekawahyu, 2018-01-03:
Ok, the slave.js example works. A quick rewrite on my code based on the example:
```javascript
function handleWriteMultipleRegistersRequest(unit, request, functionData, respond)
{
  for (i = 0; i < request.quantity; i++) {
    UNIT_TO_DATA[unit].holdingRegisters.writeUInt16BE(
      request.values.readUInt16BE(i * 2),
      ((request.startingAddress * 2) + (i * 2))
    );
  }
  request.values.copy(functionData, request.startingAddress * 2);
  respond(request);
}
```
Is there a better way to copy `request.values` into the holding registers without `for-loop`?

morkai, 2018-01-03:
That's what
```js
request.values.copy(functionData, request.startingAddress * 2);
```
is doing.
```js
functionData === UNIT_TO_DATA[unit].holdingRegisters
```

ekawahyu, 2018-01-03:
Ah, I have been trying to figure out the fix to the response and didn't notice that it was redundant! Problem solved then! I am closing this issue. Thanks.

closed 2018-01-04 by ekawahyu