morkai/h5.modbus #26: RTU Framing - Inter-frame delay/space

ekawahyu, 2018-01-05:
I have a question about the RTU framing. Which file should I look at to modify the 1.5 and 3.5 chars of inter-frame delay time?

morkai, 2018-01-05:
It's exposed in the `RtuTransport` as the [`eofTimeout`](https://github.com/morkai/h5.modbus/blob/c051b6cbce4ae6c8cc81f9989081f5b773d19c87/lib/transports/RtuTransport.js#L305) - a number of milliseconds after the last received byte after which end of frame  is assumed. It's Node.js so you have to keep in mind that `setTimeout(() => {}, 1)` might trigger after 1ms or 1s.

ekawahyu, 2018-01-05:
Yes, I understand completely that it would not be able to strictly follow the delay timing requirement by Modbus spec, but I was curious about where to look for it in case I need to modify it. Let me look at the eofTimeout.